SeppPenner/SparkplugNet #5: NBIRTH into Ignition

MRIIOT, 2021-09-29:
Great work on the library!  Have you gotten a clean NBIRTH into Ignition?  I am running into Ignition throwing exceptions and will be investigating why that is.  Will keep you updated.

MRIIOT, 2021-09-30:
I can confirm everything works as expected so far.  

Encountered issue is Cirrus Link specific: https://forum.cirrus-link.com/t/topic-segment-group-id-fails-to-parse/126/2

SeppPenner, 2021-09-30:
Can you give me some hints on how I can test this with Ignition? I have tried, but honestly didn't understand how Ignition works properly :D

MRIIOT, 2021-10-01:
Sure.

1. Install Ignition. https://inductiveautomation.com/downloads/ignition/8.1.10

2. Install CirrusLink modules. https://inductiveautomation.com/downloads/third-party-modules/8.1.10

- MQTT Distributor Module
- MQTT Engine Module
- MQTT Transmission Module 
- MQTT Recorder Module

3. Log into the gateway.  <gw-ip>:8088
4. Activate 2 hour trial
5. Configure MQTT Engine. https://docs.chariot.io/display/CLD79/ME:+Configuration

You should be able to connect to <gw-ip>:1883 MQTT broker.
You should also be able to send SpB payloads.

If you would like, we can get on a call and go through it.

-chris

MRIIOT, 2023-01-13:
Reopening this as it is related.

Observation on a working SpB NBIRTH from Ignition built-in example where `seq=0`.

```
On topic=spBv1.0/My MQTT Group/NBIRTH/Edge Node f9ace0: Incoming payload: SparkplugBPayload [timestamp=1673650361762, metrics=[Metric [name=Node Control/Next Server, alias=null, timestamp=1673650361763, dataType=Boolean, isHistorical=null, isTransient=null, metaData=null, properties=null, value=false, isNull=false], Metric [name=Node Info/Transmission Version, alias=null, timestamp=1673650361763, dataType=String, isHistorical=null, isTransient=null, metaData=null, properties=null, value=4.0.12 (b2022072818), isNull=false], Metric [name=bdSeq, alias=null, timestamp=1673650361762, dataType=Int64, isHistorical=null, isTransient=null, metaData=null, properties=null, value=0, isNull=false], Metric [name=Node Control/Rebirth, alias=null, timestamp=1673650361763, dataType=Boolean, isHistorical=null, isTransient=null, metaData=null, properties=null, value=false, isNull=false]], seq=0, uuid=null, body=null]
```

```
Processing NBIRTH from Edge Node My MQTT Group/Edge Node f9ace0 with Seq# 0
```

Here is an example of non-working SpB NBIRTH from custom client using SparkplugbNet 1.2.0 where `seq=null`.

```
On topic=spBv1.0/fanuc/NBIRTH/CSC-PF1D271K_f_sim_spb_long: Incoming payload: SparkplugBPayload [timestamp=1673650953121, metrics=[Metric [name=IpAddress, alias=null, timestamp=null, dataType=String, isHistorical=null, isTransient=null, metaData=null, properties=null, value=10.1.10.206;127.0.0.1, isNull=false], Metric [name=BDSEQ, alias=null, timestamp=null, dataType=Int64, isHistorical=null, isTransient=null, metaData=null, properties=null, value=0, isNull=false]], seq=null, uuid=null, body=null]
```

```
Invalid payload missing sequence number: spBv1.0/fanuc/NBIRTH/CSC-PF1D271K_f_sim_spb_long
```

However the same message decoded in MQTT.fx shows `seq=0`:

```
{"timestamp":1673650953121,"metrics":[{"name":"IpAddress","dataType":"String","value":"10.1.10.206;127.0.0.1"},{"name":"BDSEQ","dataType":"Int64","value":0}],"seq":0}
```

Here is a recent topic on the issue:  https://forum.cirrus-link.com/t/nbirth-keeps-getting-rejected-for-missing-sequence/244

> The SparkplugNet package is using ProtoBuf version 3, which uses the “surpress default values” feature, while the .proto definition specifies version 2, which does not.

I can confirm this observation with Wireshark.

Here is a message where `seq>0`.

Here is the NBIRTH.

MRIIOT, 2023-02-03:
>>The SparkplugNet package is using ProtoBuf version 3, which uses the “surpress default values” feature, while the .proto definition specifies version 2, which does not.

https://github.com/protocolbuffers/protobuf/issues/1606

@SeppPenner does this mean that the proto file needs to be updated?  And if so, then that means that SpB spec needs to be updated to pb3.

Kvargefar, 2023-02-03:
Hi, 
I've also encountered this issue. After a lot of troubleshooting, I'm struggling to find any other reason why Ignition rejects NBIRTH each time, other than this.

MRIIOT, 2023-02-07:
I'm looking back at the change logs and it looks like protobuf-net 3.0.73 has been used since February 16, 2021.

I guess my next step is to start downgrading and see when things actually break in Ignition.

https://github.com/SeppPenner/SparkplugNet/tags

MRIIOT, 2023-02-08:
Went to 1.0.0 but ran into a null reference with LWT message UserProperties.
Tried going back to 0.7.0, but required that I downgrade MQTTnet to 3 and I can't do that.
I'll pull the source next.

MRIIOT, 2023-02-08:
Ok that was pretty simple after getting lost for a couple hours.

[image]

https://github.com/SeppPenner/SparkplugNet/blob/f3cb22aa889e1d803a162d9acf8c9b433f7e4c5b/src/SparkplugNet/VersionB/ProtoBuf/ProtobufPayload.cs#L26-L27

Has to include `IsRequired = true`.

```
[global::ProtoBuf.ProtoMember(3, Name = @"seq", IsRequired = true)]
public ulong Seq { get; set; }
```

https://code.google.com/archive/p/protobuf-net/wikis/Attributes.wiki

Any side effects anyone can think of?

Kvargefar, 2023-02-08:
@MRIIOT did the isRequired flag help you?
I tried, but didn't find any difference in Ignition.

After closer inspection, I can see that all my payloads are accepted, except the NBIRTH. This is the first message and seq=0. Looking at the parsed result in Ignition, I can now see that seq=null. Not sure why. I've been testing different methods, but cannot a solution yet. Other clients read the first seq value as 0.

[image]

MRIIOT, 2023-02-08:
> @MRIIOT did the isRequired flag help you?
> 
> I tried, but didn't find any difference in Ignition.
> 
> Which protobuf-net version are you using. Other packages that aren't latest?
> 
> 

I did update protobuf-net to latest.

Did you add the parameter to the correct seq?

Kvargefar, 2023-02-08:
I tried adding it to member 3: 
```
[global::ProtoBuf.ProtoMember(3, Name = @"seq", IsRequired = true)]
        public ulong Seq { get; set; }
```
Okay - v3.1.33

MRIIOT, 2023-02-08:
I did the same :)

MRIIOT, 2023-02-08:
Once you test, someone should PR.

SeppPenner, 2023-02-21:
Sorry guys for the late replies. I need to check this (If someone has a clue already if the required flag is used incorrectly somehow), just tell me or add a PR if you want :)

Have been quite busy at work the last weeks...

MRIIOT, 2023-02-21:
> (If someone has a clue already if the required flag is used incorrectly somehow), just tell me or add a PR if you want :)

I need to do more testing.  I am seeing other fields besides 'seq' causing issues as well.

SeppPenner, 2023-02-23:
Since Eclipse brought out the TCK kit for testing (3.0 compatibility), I hope to get this done soon and maybe find some more issues as well...

Kvargefar, 2023-02-23:
Been away from this for a couple of weeks now. In regards to publishing/producing, I currently have good results into Ignition after changing the isRequired flag. But I have to do more testing as well.

Storm-Soft, 2023-04-06:
Hi, 
first of all thanks for this library development.

I'm also in trouble with the NBIRTH message into ignition.

After reading this exchange I've made the test of setting the "IsRequired" property on sequence as proposed.

[global::ProtoBuf.ProtoMember(3, Name = @"seq", IsRequired = true)]
public ulong Seq { get; set; }

it works like a charm.

@Kvargefar could you make some extra testing and by the way, can you plan a PR ?

Thanks

SeppPenner, 2024-03-03:
I'm now testing with the official V3.0 TCK, Please follow https://github.com/SeppPenner/SparkplugNet/issues/39 for updates.

closed 2024-03-03 by SeppPenner