SeppPenner/SparkplugNet #6: Implementing as Primary Application

MRIIOT, 2021-10-01:
What is known to be missing in order to build a SpB Primary Application on top of the library?

SeppPenner, 2021-10-04:
Well, I hope nothing to be honest :D

SeppPenner, 2021-10-08:
> Something I noticed is that the metrics have to been known ahead of time by the application. Sending new metrics from nodes, ends up in here:
> 
> https://github.com/SeppPenner/SparkplugNet/blob/ed77d8aec6489af663f9578b2e3516b82dfff293/src/SparkplugNet/Core/Application/SparkplugApplicationBase.cs#L568

I assumed that this was intended within Sparkplug...

MRIIOT, 2021-10-09:
It's probably implementation specific, but I would expect arbitrary nodes and devices to connect to the primary application without any previous knowledge.

SeppPenner, 2022-02-08:
Sorry for the late reply. I will check this within the next days and provide an update.

SeppPenner, 2022-03-13:
I will close this now, since I believe this is according to the standard. If anyone has more information or clearification, just contact me here, please :)

closed 2022-03-13 by SeppPenner

MRIIOT, 2022-03-13:
> I will close this now, since I believe this is according to the standard. If anyone has more information or clearification, just contact me here, please :)

That's fine.  Once I have more time, I will re-open this with examples.