SeppPenner/SparkplugNet #8: All SparkPlugB messages have retain flag set

emr-anthony-amaro, 2022-02-02:
Noticed this through Wireshark. NDATA, DDATA, and others have the "retain" flag set. I think it's because most of the messages are built in SparkplugMessageGenerator.cs with the .WithRetainFlag() set which has a default input parameter of TRUE. With some exceptions, most should be set to .WithRetainFlag(False).

I suspect that the SparkPlugB standard states only the STATE message should have the retain flag set. 

This is excellent work here. I hope this helps!

SeppPenner, 2022-02-06:
I guess, you're right with that. I'm not sure if it's in the standard or if I just added the retain flag on my own for all messages (because it made sense at that time).

I will check this and update the library. Already wanted to update it some time ago, but didn't have much time sadly.

SeppPenner, 2022-02-08:
> I suspect that the SparkPlugB standard states only the STATE message should have the retain flag set.

I have re-read the standard and you're right. Only the **STATE** messages of the primary application are retained.

SeppPenner, 2022-02-08:
Fixed with https://github.com/SeppPenner/SparkplugNet/commit/de9acb38bfa867319edc8846d1e8363dfe370be2. I will try to release a new version soon :)

closed 2022-02-08 by SeppPenner

SeppPenner, 2022-02-08:
Version 0.5 is on the way :)