SeppPenner/SparkplugNet #12: Encoding issue

DimitarKrastev, 2022-02-15:
Great project, I am testing out whether this could fit a client project, but I encountered a problem.

The published messages appear with unrecognized symbols when viewed in MQTT Explorer.

Any idea what might be causing this?
[image]

Environment:
.Net 6.0
Mosquitto MQTT broker
MQTT Explorer

SeppPenner, 2022-02-16:
The messages are "encoded" as binary data using [ProtoBuf](https://developers.google.com/protocol-buffers/) data model from the Sparkplug specification. Check out http://www.steves-internet-guide.com/spsparkplug-payloads-and-messages/ as well, where the guy (Steve) states that you need a special client to view the data properly (Under section "Using The MQTT tools mosqutto_pub and mosqutto_sub with Sparkplug").

Maybe I will provide a client for this in the future, too... Let's see.

SeppPenner, 2022-02-18:
Just for reference: MQTT.fx supports a Sparkplug payload decoder, but you would need to take an old version or buy a license for a new version (The old version works just fine, also with Sparkplug: https://www.jensd.de/apps/mqttfx/1.7.1/).

closed 2022-02-18 by SeppPenner