SeppPenner/SparkplugNet #13: BDSeq being sent for messages other than NBIRTH and NDEATH

emr-anthony-amaro, 2022-03-12:
As far as the Sparkplug B spec, I believe it's only for the NBIRTH and NDEATH messages. These are the places where it seems to be incorrectly added. I'm not so sure about Sparkplug A.

master/src/SparkplugNet/Core/Messages/SparkplugMessageGenerator.cs
Lines: 167,272,329,393,451, and 517

I think the fix would be to not call AddSessionNumberToMetrics() for each of those message types.

If I'm out to lunch on this please feel free to close. 

Thank you!

SeppPenner, 2022-03-13:
Yeah, that's what I wrote under https://github.com/SeppPenner/SparkplugNet#special-notes in the readme. I know, it's not in the standard, but the standard is not really satisfying either if you ask me... I can add an option flag if someone wants to use the BDSEQ in all messages, and leave it set to false i default mode.

emr-anthony-amaro, 2022-03-14:
My apologies, missed the special comment. Thanks!

closed 2022-03-14 by emr-anthony-amaro

SeppPenner, 2022-03-16:
I'm not sure if it's a good idea to not stick to the standard completely, but as far as I have seen it, it makes sense to add the BDSEQ on these messages. The Sparkplug standard is quite strange in some cases...