SeppPenner/SparkplugNet #14: No notification of received message

bcpberry, 2022-03-21:
Hi,

Firstly thanks for the effort in providing this library.

I have what I assume must be a really dumb question, but for the life of me I cannot figure out the answer. How do I receive notification of an DDATA message in an Application expecting metric updates from a Node? 

I expected to see something like an OnMessageReceived event or callback on the SparkplugApplicationBase class (similar to the OnDisconnected action), but can find nothing like this. Looking through the code, I see that when an MQTT message is received, the updated metrics are added to a concurrent dictionary, but how is my application code to know that this has occurred?

Thanks in advance for any help.

Kind regards,
Peter

SeppPenner, 2022-03-24:
> I have what I assume must be a really dumb question, but for the life of me I cannot figure out the answer. How do I receive notification of an DDATA message in an Application expecting metric updates from a Node?

I guess, it's not a dumb question 😄 This is just not yet possible as there is no action defined. I will add new events for DDATA and NODEDATA requests within the application.

SeppPenner, 2022-03-24:
With version 0.7, there is now the following options (Check https://github.com/SeppPenner/SparkplugNet/blob/master/HowToUse.md):

```csharp
// Handle the application's node data received event.
application.OnNodeDataReceived += OnVersionANodeDataReceived;

// Handle the application's device data received event.
application.OnDeviceDataReceived += OnVersionADeviceDataReceived;

private static void OnVersionANodeDataReceived(KuraMetric metric)
{
	// Do something.
}

private static void OnVersionADeviceDataReceived(KuraMetric metric)
{
	// Do something.
}
```

Just re-open the issue or create a new one if it doesn't work as expected. The new version should be on NuGet soon.

closed 2022-03-24 by SeppPenner