SeppPenner/SparkplugNet #15: Nodes cannot invoke NCMD/DCMD events

OffTravel, 2022-07-11:
In SparkplugNodeBase.cs method AddMessageReceivedHandler(), at lines 307, 317 the code is trying to convert the payload class into the metric class <T> and NCMD/DCMD events are never invoked.

```
case SparkplugNamespace.VersionB:
                            var payloadVersionB = PayloadHelper.Deserialize<VersionBProtoBuf.ProtoBufPayload>(e.ApplicationMessage.Payload);

                            if (payloadVersionB is not null)
                            {
                                var convertedPayload = PayloadConverter.ConvertVersionBPayload(payloadVersionB);

                                if (topic.Contains(SparkplugMessageType.DeviceCommand.GetDescription()))
                                {
                                    if (convertedPayload is not T convertedPayloadVersionB)
                                    {
                                        throw new InvalidCastException("The metric cast didn't work properly.");
                                    }

                                    this.DeviceCommandReceived?.Invoke(convertedPayloadVersionB);
                                }

                                if (topic.Contains(SparkplugMessageType.NodeCommand.GetDescription()))
                                {
                                    if (convertedPayload is not T convertedPayloadVersionB)
                                    {
                                        throw new InvalidCastException("The metric cast didn't work properly.");
                                    }

                                    this.NodeCommandReceived?.Invoke(convertedPayloadVersionB);
                                }
                            }

                            break;
```

closed 2022-08-03 by SeppPenner

SeppPenner, 2022-08-03:
Nice catch. I didn't see that one at first. Stupid issue, but the generics are always shitty and very likely to cause errors...

SeppPenner, 2022-08-03:
@all-contributors please add @ OffTravel for code test

SeppPenner, 2022-08-03:
@all-contributors please add @OffTravel for code test