SeppPenner/SparkplugNet #28: Bug on publishing data for devices.

emmanuelbertho, 2022-10-10:
Hello,

In the method PublishMessageForDevice in SparkplugNode.cs you are generating the payload with the known metrics of the node instead of the known metrics of the device.
That means if you don't register the metrics in both the node and the device you can't publish data for a device.

I replaced `this.KnownMetricsStorage.FilterOutgoingMetrics(metrics)` with `deviceMetrics` in the method and it is working well.

SeppPenner, 2022-10-11:
Yeah, that sounds like a bug to me. I will check it soon :)

SeppPenner, 2022-10-11:
It seems that we need two different metrics stores here.

SeppPenner, 2022-10-30:
Ah, I see. I guess, `this.KnownMetricsStorage.FilterOutgoingMetrics(deviceMetrics)` should fix the issue, correct?

emmanuelbertho, 2022-11-03:
> 

I don't think that node a should be aware of the metrics of all his devices. Well I didn't saw anything that says so in the Sparkplug B documentation. Correct me if I am wrong

Patrick2607, 2022-11-03:
> > 
> 
> I don't think that node a should be aware of the metrics of all his devices. Well I didn't saw anything that says so in the Sparkplug B documentation. Correct me if I am wrong

You might be right, I haven't found that either when I just checked, it was my understanding. However, the proposed solution in the commit is not what you intended right? In your OP you said to replace that whole line with `deviceMetrics`, but in the commit, only the parameter has been replaced like this: `this.KnownMetricsStorage.FilterOutgoingMetrics(deviceMetrics)`. This causes the issue I mentioned above. Doing it your way works, but it does not catch unknown Metrics.

I think each Device needs its own MetricStorage. Agreed?

SeppPenner, 2022-11-03:
I guess this was my mistake, the devices should have an own metrics storage for sure...

Patrick2607, 2022-11-03:
> I guess this was my mistake, the devices should have an own metrics storage for sure...

I'm working on it right now :) seems like a fairly easy fix

SeppPenner, 2022-11-08:
I've seen it, I hope to check and test it today :)

closed 2022-11-16 by SeppPenner

SeppPenner, 2022-11-20:
New version available soon :)

PR #31 (2022-11-16): Device Storage
- Changed KnownDevices to use MetricStorage instead of a List of Metrics
- Refactored SparkplugNodeBase.PublishMessageForDevice
- Fixed an assignment issue where the DeviceIdentifier would represent itself as the EdgeNodeIdentifier in the DeviceDataEventArgs

This should resolve #28