SeppPenner/SparkplugNet #30: SparkplugNet.Core.PayloadConverter.ConvertVersionBPayload discards byte[]

49564B0F, 2022-10-27:
Hi,
The SparkplugNet.Core.PayloadConverter.ConvertVersionBPayload seems to remove data if the data type is Bytes.
Here is a snippet to reproduce the issue:
```
var payload = new SparkplugNet.VersionB.Data.Payload
{
	Metrics = new Metric[] { new SparkplugNet.VersionB.Data.Metric()
	{
			BytesValue = new byte[]{1,2,3,4}
	}
	}.ToList(),
	Seq = (ulong)1,
	Timestamp = (ulong)DateTimeOffset.Now.ToUnixTimeMilliseconds()
};

var payloadConverter = typeof(SparkplugNet.VersionB.SparkplugNode).Assembly.GetType("SparkplugNet.Core.PayloadConverter");
var convertMetric = payloadConverter.GetMethod("ConvertVersionBPayload", new Type[] { typeof(SparkplugNet.VersionB.Data.Payload) });
var result = convertMetric.Invoke(null, new object[] { payload });
var metrics = result.GetType().GetProperty("Metrics").GetValue(result);
var metric = ((IList)metrics)[0];
var byteValues = metric.GetType().GetProperty("BytesValue").GetValue(metric);
var dataType = metric.GetType().GetProperty("Datatype").GetValue(metric);
Console.WriteLine($"{dataType} : Items={string.Join(",", ((byte[])byteValues))} Length={((byte[])byteValues).Length}" );
```

SeppPenner, 2022-10-27:
Thanks for the test code, I will need to check this.

SeppPenner, 2022-10-30:
Confirmed, this is a bug. I will try to fix it now.

SeppPenner, 2022-10-30:
This is fixed in the next version, I hope to provide it soon.

closed 2022-10-30 by SeppPenner