SeppPenner/SparkplugNet #33: Not getting BIRTH message into the  SparkplugNet.Examples  when a new publisher comes up on the broker

venkat9200, 2022-12-02:
Hi SeepPenner.
I'm using the SparkPlugNet samples and running RunVersionB()  function.
Application class is able to receive the BIRTH messages from RunVersionBNode() for group1 and device 1.

On the same broker address I have another publisher publishing data at Group Name : "Group" and NodeName: "EdgeGateway" and devideName : "device_0"

When this publisher goes down Application class is able to receive the DEATH messages. 
But when this publisher comes up application is NOT able to receive the BIRTH messages.

Could you please let me know if any changes need to be done inside the RunVersionBApplication() OR in RunVersionBNode().
I verified that the topic it is subscribing to is spBv1.0/#  , but birth messages from another publisher is not able to detect.
Whereas MQTT.fx application is able to detect the BIRTH messages from the broker address.
I'm submitting this as an issue as I  do not see a discussion board here , please don't for this.

Thanks in Advance
Venkat

SeppPenner, 2022-12-23:
Sounds strange, I need to investigate. Maybe a callback from MqttNet is not triggered properly...

evonz-mx, 2023-02-22:
I'm having a similar problem with my spBv1.0 application, which will include both Node and Application components. My application gets death messages, but never gets birth or data messages.  However, using debug logging on my broker shows that all the missed messages are delivered if I restart my application (but before the app is ready to receive messages, it seems).    Third party applications like MQTT Explorer or MQTT.fx get all the messages in realtime.

I'm guessing that the problem is QoS: the SparkplugNet subscriptions are with QoS 1, but are never acked explicitly.  I don't know if MQTTNet defaults autoack to true, but if not, that would explain my problem.  I haven't tried implementing that in your code yet, but I figured I'd drop this note in case you get to it before I do.  If I find success with this approach, I'll send a PR.

evonz-mx, 2023-02-22:
Just a followup to my own comment in case anyone else hits similar issues: it doesn't look lke QoS is my issue.  Instead, it looks like metric storage is throwing an exception since the devices are publishing metrics that haven't yet been seen.  My application can't know all possible metrics ahead of time (it connects to unknown devices to allow users to select and log metrics), so I'm going to move towards building a mechanism to request a rebirth in this scenario, and allow "learning" metrics from birth messages.

SeppPenner, 2023-04-07:
So, this seems to be the same issue as https://github.com/SeppPenner/SparkplugNet/issues/43. The problem is that this is currently by design in version 2.2: https://github.com/SeppPenner/SparkplugNet/issues/6#issuecomment-1044441756. I will check version 3 compatibility today or tomorrow and check if some changes have been done there. Until then follow https://github.com/SeppPenner/SparkplugNet/issues/43 for updates.

closed 2023-04-07 by SeppPenner