SeppPenner/SparkplugNet #37: Empty Metric Names in NDATA

michael-benoit, 2023-02-07:
Don't know if this a bug or not but I'm trying to publish xDATA messages with blank names. If the name is blank or doesn't match the xBIRTH message. The metrics array in the payload ends up being empty.

Names are not required as part of the Sparkplug B spec. 
Excerpt:
"Upon being defined in the NBIRTH or DBIRTH, subsequent messages can supply only the alias instead of 
the metric friendly name to reduce overall message size." 

TIA

SeppPenner, 2023-02-21:
Sounds like a bug (More or less something I used to overlook in the docs as I was just to eager about nullable checks and that stuff :D)

SeppPenner, 2023-04-07:
I read through the 3.0 spec and there are some more rules that seem to apply in this case:

▪ [tck-id-payloads-alias-birth-requirement] NBIRTH and DBIRTH messages MUST include both a metric name and alias.
▪ [tck-id-payloads-alias-data-cmd-requirement] NDATA, DDATA, NCMD, and DCMD messages MUST only include an alias and the metric name MUST be excluded.

So, to summarize this for me: If a metrics alias is used, the metric name and the metric alias must be set at birth and only the alias must be set at all other messages making it invalid if a metric name is set there.

lsmythe, 2024-02-02:
Has there been any progress made on this issue? Keen for an update :)

SeppPenner, 2024-02-10:
> Has there been any progress made on this issue? Keen for an update :)

Sorry, not yet. I try to work on the library today and tomorrow to fix the most crucial things...

SeppPenner, 2024-03-14:
I have already added this in my current debugging state of the library, will check to fix this soon.

closed 2024-03-20 by SeppPenner