SeppPenner/SparkplugNet #38: Question: QoS

michael-benoit, 2023-02-13:
How do I specify the QoS of a message to publish?

SeppPenner, 2023-02-21:
Good question, I need to check this again. Still some time after my last changes here...

SeppPenner, 2023-02-23:
The short answer at the moment is: You can't :D Don't ask me why, but somehow I assumed that all Sparkplug messages should be sent with QoS=1 (Which in 2.2 was only the case for state messages as far as I researched it now). I will try to change this and provide full 3.0 compatibility with the TCK from Ecplise as requested under https://github.com/SeppPenner/SparkplugNet/issues/39, so I guess you need to wait for that to be done.

SeppPenner, 2023-04-07:
Needs to be checked. Some messages must be sent with QoS 0 and retain = false (E.g. NBIRTH, NDATA).

SeppPenner, 2023-04-07:
Is checked against Sparkplug version 3.0:

NBIRTH --> QoS = 0
NDEATH --> Not fix
DBIRTH --> QoS = 0
DDEATH --> QoS = 0
NDATA --> QoS = 0
DDATA --> QoS = 0
NCMD --> QoS = 0
DCMD --> QoS = 0
STATE --> QoS = 1

Only NDEATH allows to specify a QoS explicitely.

SeppPenner, 2023-04-08:
Updated: Completely invalid:

* Edge node will message QoS must be set to `1`.
* Edge node will message retain flag must be set to `false`.

closed 2023-04-08 by SeppPenner