SeppPenner/SparkplugNet #39: Eclipse™ Sparkplug™ Technology Compatibility Kit (TCK)

BoBiene, 2023-02-15:
Have any of you already investigated TCK, and by any chance is SparkplugNet already tested for compatibility?

https://github.com/eclipse-sparkplug/sparkplug/blob/develop/tck/UserGuide.adoc
https://sparkplug.eclipse.org/compatibility/get-listed/

SeppPenner, 2023-02-21:
Same answer as to the 3.0 compatibility. No yet. Will check this soon as well.

scottmwyant, 2023-09-11:
Taking initial steps to look into this.  Have not worked with TCK before, so I'm probably not going to be super effective here.  Looking to write SpB "clients" and this library + tck seem to be a good start.

- eclipse-sparkplug/sparkplug#482
- eclipse-sparkplug/sparkplug#484

SeppPenner, 2023-12-07:
I have NVM running on Windows (https://github.com/coreybutler/nvm-windows), version 1.1.12, node version 21.4.0 and tried to follow https://github.com/eclipse-sparkplug/sparkplug/blob/develop/tck/UserGuide.adoc:

```bash
npm install -g yarn
corepack enable
yarn install
```

throws some errors with `yarn install` (AND again I know why I hate all that node.js shit).
Logfile can be found under https://github.com/SeppPenner/YourMotherIsALogFile/blob/master/Logfile.md (Because of the 65536 character limit in comments...)

[image]

Does anyone have an idea here?

scottmwyant, 2023-12-07:
You might need a flag: `--openssl-legacy-provider`. See bullet two in my earlier comment.

shouidar, 2024-01-29:
run the tck test on Fedora vm successfully. Runing with the original code most tests show not executed. 
See atteched:
[results1.txt](https://github.com/SeppPenner/SparkplugNet/files/14088744/results1.txt)

Version 3 of the spec requires the Host Application STATE birth message topic to be: "spBv1.0/STATE/HostId", running the test with this hard coded into the library executes all tests with some failing due to other aspects of the 3.0 spec.
See attached: [results2.txt](https://github.com/SeppPenner/SparkplugNet/files/14088746/results2.txt)

scottmwyant, 2024-02-05:
@SeppPenner I got the [ERR_STREAM_PREMATURE_CLOSE] as you did.  Running `yarn set version latest` resolved the problem for me.  I didn't dig in to understand the problem here, but seems related to *corepack* subsystem and the version of *yarn* itself.

I still ran into the issue I mentioned above at the `yarn build` step, but using the `--openssl-legacy-provider` option did resolved it. 
 Reference: https://stackoverflow.com/a/69699772

From there I was able to `yarn start` and get to the tck at `http://localhost:3000`.

SeppPenner, 2024-02-10:
> @SeppPenner I got the [ERR_STREAM_PREMATURE_CLOSE] as you did. Running `yarn set version latest` resolved the problem for me. I didn't dig in to understand the problem here, but seems related to _corepack_ subsystem and the version of _yarn_ itself.
> 
> I still ran into the issue I mentioned above at the `yarn build` step, but using the `--openssl-legacy-provider` option resolved it. Reference: https://stackoverflow.com/a/69699772
> 
> From there I was able to `yarn start` and get to the web console at `http://localhost:3000`.

Thanks, I have seen this already, but didn't yet come to check it... There is still a lot of stuff to do to get 3.0 compatibility, I guess...

BoBiene, 2024-02-10:
The most recent version of the TCK is on the development branch i think:
https://github.com/eclipse-sparkplug/sparkplug/tree/develop/tck