SeppPenner/SparkplugNet #44: Unable to send NBIRTH Message after Primary App returns ONLINE

michael-benoit, 2023-03-04:
Need the ability to send a NBIRTH message similar to PublishDeviceBirthMessage.

When the Primary App returns to ONLINE, stored messages need to be sent after a new NBIRTH message is sent.

Kvargefar, 2023-03-15:
I need this as well. I currently have to use Device Control/Rebirth.

SeppPenner, 2023-04-07:
I need to check for Sparkplug 3.0 conformity first (Will be the next step, I hope to do it today or tomorrow) and the next thing is to check this request.

Kvargefar, 2023-05-05:
@SeppPenner I believe this functionality is needed to comply with the 2.2 standard. I haven't checked the 3.0 standard. In the meantime, I believe this is solved by making the Node.PublishInternal() method public. The node should however be able to distinguish between the node metrics and device metrics - which I'm looking into now.

buentead, 2023-06-27:
In Sparkplug v2.2 specification, the Edge Node should be able to respond to the command _Node control/Rebirth_ (chapter 17.5, page 66). According to the Sparkplug v3.0.0 specifications the Edge Node is required to implement the _Node control/Rebirth_ command (chapter 5.17, page 56, last paragraph on page). As of now, I could only implement this command if I Stop/Start the Edge Node connection - and this doesn't really make sense, does it.
As the command _Node control/Rebirth_ is a MUST have in Sparkplug v3.0.0, would it make sense that the SparkplugNet library handles this command itself?
I would really appreciate a solution soon, as we need to be able to handle the Rebirth command within the Edge Node. Thanks.

adityashahazilen, 2023-11-07:
@SeppPenner Do you have any update on this feature?

SeppPenner, 2024-03-24:
So, the main idea here is to have a method to do a node rebirth that (in contrary to a `Stop()` and `Start()` of a node) does only send a NDEATH and a new NBIRTH (without doing anything MQTT specific like disconeccting, reconnecting and subscribing, etc.). Or do I understand this in a wrong way?

closed 2024-03-27 by SeppPenner