SeppPenner/SparkplugNet #48: FilterIncomingMetrics instead of ValidateIncomingMetrics?

Patrick2607, 2023-05-02:
In **src/SparkplugNet/Core/SparkplugBase.KnownMetricStorage.cs**, there is the following piece of code that checks whether an incoming message ONLY contains the metrics that are known, else it will throw an exception.

```c#
public virtual void ValidateIncomingMetrics(IEnumerable<T> metrics)
{
    foreach (var metric in metrics.Where(metric => !this.ContainsKey(metric.Name)))
    {
        throw new Exception($"Metric {metric.Name} is an unknown metric.");
    }
}
``` 

However, in my case, I'm only interested in SOME tags that get send, and I want them to ignore the rest. For instance, let's say I'm interested in the following device metrics:

'temperature', DataType.Float
'enabled', DataType.Boolean

But the device sends the following data:

'temperature', DataType.Float
'pressure', DataType.Float
'count', DataType.Int32
'enabled', DataType.Boolean

Configuring the metrics with only the 'temperature' and 'enabled' will result in a thrown exception. This is unwanted behaviour from my side. Same goes for the Node, I am not interested in the Node data, however, when I don't add the metrics to the known list, it will throw an exception as well. I'd rather have them filtered; similarly to the **FilterOutgoingMetrics** function in **SparkplugBase.KnownMetricStorage.cs**. The following function solved my problem:

```c#
public virtual IEnumerable<T> FilterIncomingMetrics(IEnumerable<T> metrics)
    => metrics.Where(metric => this.ContainsKey(metric.Name));
```

Is there a specific reason an exception is thrown in this case? If so, how would I be able to resolve this issue? If not, shall I create a pull request that has an option to validate or filter the incoming metrics?

Patrick2607, 2023-05-09:
I thought I'd help out and created a possible solution (see #50)

SeppPenner, 2024-03-14:
I agree now, your proposal to filter the metrics makes more sense. I already work on this. I really try to provide that feature ASAP...

closed 2024-03-21 by SeppPenner