SeppPenner/SparkplugNet #52: Connection with Web socket

emmanuelbertho, 2023-06-15:
Hello, 

I am trying to connect to a broker with web socket and I think there is an error in the connection method in SparkplugNodeBase.cs line 296.

I think in `builder.WithWebSocketServer(this.Options.BrokerAddress, this.Options.WebSocketParameters);` the first parameter shouldn't be just the broker address but `{brokerAdress}:{port}`.

I tested the modification and it works on my side.

Emmanuel

SeppPenner, 2023-11-28:
I'm not 100 percent sure, this depends on how you want to reach your websocket, e.g. if there is a connection like `mqtt.eclipseprojects.io:443/mqtt`, this would not work. Maybe I can add something like this:

```csharp
if (this.Options.BrokerAddress is null)
{
    builder.WithWebSocketServer(this.Options.BrokerAddress, this.Options.WebSocketParameters);
}
else
{
    builder.WithWebSocketServer($"{this.Options.BrokerAddress}:{this.Options.BrokerPort}", this.Options.WebSocketParameters);
}
```

SeppPenner, 2023-12-07:
> I'm not 100 percent sure, this depends on how you want to reach your websocket, e.g. if there is a connection like `mqtt.eclipseprojects.io:443/mqtt`, this would not work. Maybe I can add something like this:
> 
> ```cs
> if (this.Options.BrokerPort is null)
> {
>     builder.WithWebSocketServer(this.Options.BrokerAddress, this.Options.WebSocketParameters);
> }
> else
> {
>     builder.WithWebSocketServer($"{this.Options.BrokerAddress}:{this.Options.BrokerPort}", this.Options.WebSocketParameters);
> }
> ```

This doesn't work as I want to keep 1883 as default TCP port...

I have added a new hint here: https://github.com/SeppPenner/SparkplugNet/blob/master/HowToUse.md#connection-via-websocket-to-mqtt-server. Since I want to keep 1883 as default port for TCP (and I'm not sure how I should handle the WebSocket scenario with that precondition, my recommendation would be to specify the complete url (e.g. mqtt.eclipseprojects.io:443/mqtt or maybe test.something.com:443) as option for BrokerAddress.

closed 2023-12-07 by SeppPenner