SeppPenner/SparkplugNet #53: Unable to call NBIRTH payload generate method explicitly

DarshitGandhi-net, 2023-06-23:
There is functionality in sparkplug specification where you can send NBIRTH message only if Host Application is online. So when Primary Host Application is online it will send STATE payload with online message after that you can send stored message in NBIRTH payload but in this Node.PublishInternal method for generating NBIRTH message of SparkPlugBase class it's directly publishing message without verifying Host app is online or not . Need this functionality for this .

ankitagojiya, 2024-02-29:
Hello @SeppPenner ,
Any news on this ?

SeppPenner, 2024-02-29:
Sorry, no. I'm currently working on the payload conversion. And I want to finish this finally (Yes, I already wrote that several times) within the next 3 to 4 days...

ankitagojiya, 2024-03-14:
Hello @SeppPenner 
Apologies for the repeated inquiries, but could you please provide an update on this?

Because if I want to send NBIRTH second time from already running application i need to call this Start("await node.Start(nodeOptions)") method because we don't have any explicit method to send NBIRTH and "Start" method will internally reconnect and subscribe topics unnecessary and also we have to subscribe to NCMD and STATE event also because it has established new connection so we won't get any STATE and NCMD message if we don't subscribe events again. It is good to make "await this.PublishInternal()" public or explore alternative solutions
This issue is impeding current implementations. 

Related issue : https://github.com/SeppPenner/SparkplugNet/issues/44

[image]