SeppPenner/SparkplugNet #54: Array DataType Support

adityashahazilen, 2023-06-29:
Following  data types are not supported  in package

   Int8Array = 22;
  Int16Array = 23;
  Int32Array = 24;

  Int64Array = 25;
  UInt8Array = 26;
  UInt16Array = 27;
  UInt32Array = 28;
  UInt64Array = 29;
  FloatArray = 30;
  DoubleArray = 31;
  BooleanArray = 32;
  StringArray = 33;
  DateTimeArray = 34;

SeppPenner, 2023-11-28:
Thank you, need to check this.

SeppPenner, 2023-11-30:
I have added the data types, but I'm stuck with the conversion somehow for now... I used https://protogen.marcgravell.com/ for the proto data from the specs to generate a C# class which worked very well. However, the conversion misses `Details`, `ValueOneofCase`, `ValueCase` and `Extensions`. Does anyone have an idea here? I will add the (not working) conversion for now.

SeppPenner, 2023-12-18:
Should be fixed after https://github.com/SeppPenner/SparkplugNet/commit/4fdb44934b65764d2d589e7ac5b4fab9b7474a93 (Version 1.3.0+).

closed 2023-12-18 by SeppPenner