SeppPenner/SparkplugNet #55: Datetime metric types

evonz-mx, 2023-07-21:
Hi folks,

When writing a DateTime value, the sparkplug data type is converted to Int64 (4).  But I'm hoping to keep the datatype as DateTime (13).  I think this is happening because ValueBaseVersionB.SetValue sets the value via .LongValue, but doesn't preserve the type.

Is this intended?  Should I be overriding the metric's type after instantiation (as I am currently doing to work around this issue)?

SeppPenner, 2023-12-07:
Will be done, for my reference https://github.com/eclipse/tahu/blob/5736e404889d4b95910613040a99ba79589ffb13/java/lib/core/src/main/java/org/eclipse/tahu/message/SparkplugBPayloadDecoder.java and https://github.com/eclipse/tahu/blob/5736e404889d4b95910613040a99ba79589ffb13/java/lib/core/src/main/java/org/eclipse/tahu/message/SparkplugBPayloadEncoder.java.

SeppPenner, 2023-12-18:
Should be fixed after https://github.com/SeppPenner/SparkplugNet/commit/4fdb44934b65764d2d589e7ac5b4fab9b7474a93 (Version 1.3.0+).

closed 2023-12-18 by SeppPenner

evonz-mx, 2023-12-18:
Thanks very much!  I look forward to trying this fix out in our application.

SeppPenner, 2023-12-18:
I'm not yet sure if this works properly already. I still need to iplement more serialization Tests here...