SeppPenner/SparkplugNet #56: Metric Class Integer Property is uint?

Kvargefar, 2023-08-07:
Is there a reason why the integer property is an unsigned integer? I've encountered several exceptions when creating a new instance of a Metric, or using the SetValue method, when the value is a negative integer.

SeppPenner, 2023-11-28:
Good question, I need to check that again. From the first view, uint doesn't make sense...

SeppPenner, 2023-12-07:
Waiting on https://github.com/eclipse-sparkplug/sparkplug/issues/495  and / or https://github.com/eclipse-sparkplug/sparkplug/issues/499.

SeppPenner, 2023-12-07:
Will be done, for my reference https://github.com/eclipse/tahu/blob/5736e404889d4b95910613040a99ba79589ffb13/java/lib/core/src/main/java/org/eclipse/tahu/message/SparkplugBPayloadDecoder.java and https://github.com/eclipse/tahu/blob/5736e404889d4b95910613040a99ba79589ffb13/java/lib/core/src/main/java/org/eclipse/tahu/message/SparkplugBPayloadEncoder.java.

SeppPenner, 2023-12-18:
Should be fixed after https://github.com/SeppPenner/SparkplugNet/commit/4fdb44934b65764d2d589e7ac5b4fab9b7474a93 (Version 1.3.0+).

closed 2023-12-18 by SeppPenner