SeppPenner/SparkplugNet #57: Must NBIRTH sequence number always be '0'?

buentead, 2023-11-27:
First I'd like to thank all contributors for this helpful library. Great job.

For me it is not quite clear, if the NBIRTH sequence number must always be '0' as it is currently implemented. The SparkplugB 3.0.0 documentation is ambiguous. In chapter 6.4.19 the sequence number "MUST have a value between 0 and 255". But in chapter 12.13. it is written "The NBIRTH MUST include a sequence number in the payload and it MUST have a value of 0."

The following theoretical example shows wrong data if the NBIRTH sequence number is always '0':
Let's assume, the 'bdseq' remains the same. The following messages are sent:
-> NBIRTH (seq=0)
-> NDATA (seq=1)
<- receiving NCMD "Node Control/Rebirth"
-> NBIRTH (seq=0)
Now let's assume the host application receives the 2nd NBIRTH message before the NDATA message. As a result, the host application would process the NDATA message as the sequence number is higher than the sequence number of the 2nd NBIRTH. This may result in wrong data. Unless the timestamp is taken into account as well.

Hence may question: what is the correct behavior of NBIRTH sequence number from your point of view? Thank you very much for any thoughts.

jeff-pf, 2023-11-28:
Also seems to contradict section 5.4 - NBIRTH must have a seq number that is between 0 and 255 (inclusive). The next paragraph says - This will become the starting sequence number which all following messages will include a sequence number that is one or more than the previous up to 255 where it wraps back to zero.

buentead, 2023-12-06:
Yes. When looking at the sequence number only, the (old) NDATA would be correct. And if a new NDATA will be sent, it would have the sequence number '1'. So the application will ignore it as the 'old' NDATA with sequence '1' was processed.

SeppPenner, 2024-03-22:
I'm not sure either. Please follow https://github.com/eclipse/tahu/issues/373 as the team (hopefully) might be able to answer this finally...

closed 2024-03-22 by SeppPenner