SeppPenner/SparkplugNet #58: SparkplugApplication Required Known Metrics

ankitagojiya, 2023-12-05:
According to sparkplug specificaton there is no need for host application to know what kind of metrics can be received from edge node . 
But according to SparkplugNet package implementation predefined NBIRTH/Metrics in the constructor must be given to start the host application. Ideally, Host application should be started without know metrics.

**Line 26 : [SparkplugNet/src/SparkplugNet/VersionB/SparkplugApplication.cs at master · SeppPenner/SparkplugNet · GitHub](https://github.com/SeppPenner/SparkplugNet/blob/master/src/SparkplugNet/VersionB/SparkplugApplication.cs)**

[image]

SeppPenner, 2023-12-07:
I guess, you're right here. Need to check it though...

ankitagojiya, 2023-12-13:
@SeppPenner 
Any updates on this ?

buentead, 2024-01-05:
@SeppPenner : I'm also interested in this fix. Do you have any timeframe when this will be fixed? Thanks.

mikolajwalus, 2024-01-05:
Same here, we would like to use this for our projekt to work as primary host scada but for work application needs to know all the metrics from the start which is a no go. This fix also looks reasonable  https://github.com/SeppPenner/SparkplugNet/pull/50/files#diff-b1c3e69c5f877405f006317b3e1cc8a2c89e9d89cb94979c339d4edf9ab28739

SeppPenner, 2024-01-05:
I'm honestly not yet sure what is correct here, but I will remove this for now.

My main problem is that I didn't have the time to check again on V3 (And that the last time I checked, the Testkit from Eclipse didn't want to start properly as yarn had issues)...

closed 2024-01-05 by SeppPenner

SeppPenner, 2024-01-05:
Version 1.3.1 is coming soon where this is removed (I still want to review this once I'm able to...)

commit 76cc3642 (2024-01-05): Merge pull request #61 from adityashahazilen/master

Bug Fix #58 - SparkplugApplication Required Known Metrics