SeppPenner/SparkplugNet #64: ArrayValues are missing metrics

ankitagojiya, 2024-01-31:
Hello @SeppPenner 

As per issue [https://github.com/SeppPenner/SparkplugNet/issues/54](https://github.com/SeppPenner/SparkplugNet/issues/54)  , I think you have added the array datatypes but its related values are still missing in metrics , pls refer below image which don't have any **_StringArrayValue_** or any array values

[image]

SeppPenner, 2024-02-09:
This is mainly the same issue as https://github.com/SeppPenner/SparkplugNet/issues/63, I guess.

SeppPenner, 2024-02-09:
Update is planned for tomorrow.

SeppPenner, 2024-02-10:
Version 1.3.2 is out soon and should fix this.

ankitagojiya, 2024-02-14:
Hello @SeppPenner , That's not available in version 1.3.2.

I mean if you are giving support of string array than I am able to set the value of string array.

For e.g. If I select string data type then I will add value in StringValue property. Same way If I select datatype as StringArray on which property I have to filed that value?

Example of string array value : ["Test1","Test2"]

shouidar, 2024-02-14:
Looks like all the Array data types are not implemented yet. @SeppPenner please confirm.

ankitagojiya, 2024-02-14:
Yes , Also please can you reopen the issue if it's not implemented yet.

SeppPenner, 2024-02-14:
I guess, you're right. I am currently testing with a broad unit test to find all issues with serialization here... Might take some time though. I hope to provide a solution until Saturday...

ankitagojiya, 2024-02-14:
Appreciate it , Thank you

SeppPenner, 2024-02-25:
I'm nearly done now. Some tests still fail because of issues with signed / unsigned values.

malavvvakharia, 2024-03-01:
@SeppPenner, do you have a specific time for releasing this issue? I'm blocked by a bug.

SeppPenner, 2024-03-03:
Fixed with https://github.com/SeppPenner/SparkplugNet/releases/tag/1.3.3.

closed 2024-03-03 by SeppPenner