SeppPenner/SparkplugNet #65: Metric Values sent with default values instead of preassigned values.

shouidar, 2024-02-02:
All Metrics sent with NDATA, DDATA, NBIRTH, DBIRTH sent with values set to default (0 for float and false for boolean) as reported by Mqtt.Fx.

Looks like there is an issue with protobuf.

[image]

### To reproduce:
Cloned the project in Visual Studio 2022 and run the example application. Monitored  pubished message with Mqtt.Fx.

shouidar, 2024-02-04:
I think the problem is in ConvertVersionBMetric, assigning several datatypes during the Metric object creation will cause the internal Discriminated union to default to the last assignment independently of the Datatype property. 
See Watch values in the screenshot below.
[image]

shouidar, 2024-02-06:
Just noticed this is beeing addressed in issue #63.
Closing this topic.

closed 2024-02-06 by shouidar

SeppPenner, 2024-02-09:
> Just noticed this is beeing addressed in issue #63. Closing this topic.

Thank you, yes, this is the same issue.