SeppPenner/SparkplugNet #66: Not able to publish metrics with values from edge node (For all datatypes)

ankitagojiya, 2024-02-09:
Hello @SeppPenner ,

Facing issue while publishing metrics with any kind of values (like.. Int , string , dataset) , at the time of converting payload into VersionBPayload (Image 1) it is overriding all the values (Image 2) that's why only the last datatype which is template (image 2) is working in version 1.3.1 , maybe we have to break the process after assigning values to any type of metric datatype which was working fine in version 1.2.0 (Image 3)

Please refer to all images below

Image 1
[image]

Image 2
[image]

[image]

Image 3 (According to version 1.2.0)
[image]
[image]

shouidar, 2024-02-09:
see issue #63

SeppPenner, 2024-02-09:
This is mainly the same issue as https://github.com/SeppPenner/SparkplugNet/issues/63, I guess.

SeppPenner, 2024-02-10:
Version 1.3.2 is out soon and should fix this.

closed 2024-02-10 by SeppPenner