SeppPenner/SparkplugNet #68: SparkPlugB 3.0 Specification primary host STATE message Payload and Topic

shouidar, 2024-02-10:
Just wanted to share some of my findings while trying to implement a Primary Host compliant with 3.0 Specification:
 
1. In the current version of the code The payload returned by *GetSparkPlugStateMessage* defaults to *UtcNow*, while in the 3.0 Specification, The application may need to send different timestamp in the State message mainly when republishing the Primary Host BIRTH message:

```C#
private static string GetSparkplugStateMessage(bool online)
    {
        return JsonSerializer.Serialize(new StateMessage
        {
            Online = online,
            Timestamp = DateTimeOffset.UtcNow.ToUnixTimeMilliseconds()
        });
    }
```

2. The sparkPlugB topic for the WILL/DEATH/BIRTH state message should be *spBv1.0/STATE/Host_Id* instead of *STATE/Host_Id*
3.  Should the current version of the library maintain backward compatibility with previous SparkPlug  B specification (1.2.1 and below) as an option?

SeppPenner, 2024-02-10:
1. I need to check this.
2. Ok, need to check this as well.
3. I honestly would avoid this unless someone really needs it...

shouidar, 2024-02-10:
For (3) I actually meant Spec version 2.2 and below.
The current implementation of *GetSparkplugStateMessageB* returns the STATE payload in JSON format as specified in 3.0. This breaks compatibility with 2.2 and below which uses UTF8 string "ONLINE" or "OFLINE" for STATE message in addition the state topics being different.

SeppPenner, 2024-02-13:
> For (3) I actually meant Spec version 2.2 and below. The current implementation of _GetSparkplugStateMessageB_ returns the STATE payload in JSON format as specified in 3.0. This breaks compatibility [...]

I will include this soon. Seems like a legit thing to do so. Especially, as it's not a big thing as only the state message has changed (For now).

SeppPenner, 2024-02-20:
Issue 3 is done, 1 and 2 need still to be checked.

SeppPenner, 2024-02-23:
2 and 3 are done, need to check 1.