SeppPenner/SparkplugNet #70: UInt64 reading problem

petrosinoE80, 2024-02-16:
First, I would like to express my gratitude to all contributors for their invaluable work to this library. 

I would like to report an issue I am experiencing during the transmission of UInt64 type metrics. When attempting to set the maximum value as follows:

```cs
metric.SetValue(DataType.UInt64, UInt64.MaxValue);
```

the DataType is instead set to Int64, and I receive a metric of the following type:

```json
"metric": [
{
"name": "metricName",
"alias": 0,
"timestamp": 1708076089783,
"datatype": 4,
"is_historical": false,
"is_transient": false,
"is_null": false,
"metadata": null,
"properties": null,
"long_value": 18446744073709552000
},
...
```

During reading, the client expects to read a variable of type Int64 ("datatype": 4), which is incompatible with the "long_value" that the metric is intended to represent at that moment. It seems that LongValue is always represented by an Int64 (rather than also being able to be UInt64).

I hope I have explained the issue clearly, and at this point, I would like to inquire if there are any existing workarounds or if I am overlooking something. Thank you very much!"

SeppPenner, 2024-02-16:
This is an issue still, yes. I hope to fix all these shitty issues until tomorrow. The latent code on master already has tests prepared already.

SeppPenner, 2024-02-25:
This problem is discussed in https://github.com/eclipse-sparkplug/sparkplug/issues/114. It seems, like we need to cast / convert the values explicitely in V3.0 (Which I don't think makes sense, but will not work else)...

SeppPenner, 2024-03-03:
Fixed with https://github.com/SeppPenner/SparkplugNet/releases/tag/1.3.3.

closed 2024-03-03 by SeppPenner