SeppPenner/SparkplugNet #72: VersionB PropertyValue.ValueCase return 20 after setting DataType to DataType.PropertySetList

thuyenle1999, 2024-02-23:
I would like to report an issue when setting VersionB PropertyValue's DataType to DataType.PropertySetList. 

```csharp
        var propertyValue = new PropertyValue()
        {
            DataType = DataType.PropertySetList, 
            PropertySetsValue = new PropertySetList()
        };
```

When I checked the ValueCase of the PropertyValue, it somehow returned 20, which indicates PropertySet, rather than 21, which indicates PropertySetList.

I can force the ValueCase to 21, but I think it should be set to 21 automatically.

I hope I have explained the problem clearly as English isn't my mother tongue.

Thank you very much for your useful library!

SeppPenner, 2024-02-23:
I'm currently reworking the total conversion and value assignment and try to back all those things by unit tests. Stay tuned, I hope to finish this soon.

SeppPenner, 2024-03-03:
Fixed with https://github.com/SeppPenner/SparkplugNet/releases/tag/1.3.3.

closed 2024-03-03 by SeppPenner