SeppPenner/SparkplugNet #73: NCDM Event only support single metric

adityashahazilen, 2024-02-26:
As per Sparkplug specification "Multiple metrics can be supplied in a single NCMD message". This means when Host Application publish NCMD with multiple metrics then Edge Node should receive all metrics in the single event. At the moment, NodeCommandReceived event only receive single metrics as a argument.

**Existing Code**
[image]

It should be like below to handle multiple metrics
[image]

shouidar, 2024-02-26:
The _SparkPlugNode_ rises an event for each Metric received in the NCMD message:
```c#
case SparkplugMessageType.NodeCommand:
                    foreach (var metric in convertedPayloadVersionB.Metrics)
                    {
                        if (metric is VersionBData.Metric convertedMetric)
                        {
                            await this.FireNodeCommandReceivedAsync(convertedMetric);
                        }
                    }

                    break;
```
But I do prefer the idea of a single event with all the metrics.

SeppPenner, 2024-02-29:
I guess, we can do that. Seems legit.

closed 2024-03-03 by SeppPenner