SeppPenner/SparkplugNet #75: NDEATH trigger problem

petrosinoE80, 2024-02-27:
Hi, I have an issue with the trigger of the death message publication. 

When I call the SparkplugBNode Stop() method, I don't receive any death certificate at the client side. 

Is there a way to trigger it forcibly, e.g. in order to call it in a Dispose() method or similar? 

[SparkplugNet ver. 1.3.2]

Thank you very much in advance.

shouidar, 2024-02-28:
When the stop method is called an MQTT Disconnect packet is sent to the broker. In this case the broker will not deliver the WILL message to subscribers. Normally the Edge Node should publish an NDEATH message before intentionally disconnecting. Unfortunately, it seems the library does not implement a method for sending DEATH message. (This is also true for the Host application).

SeppPenner, 2024-02-29:
> When the stop method is called an MQTT Disconnect packet is sent to the broker. In this case the broker will not deliver the WILL message to subscribers. Normally the Edge Node should publish an NDEATH message before intentionally disconnecting. Unfortunately, it seems the library does not implement a method for sending DEATH message. (This is also true for the Host application). This is a requirement for version 3.0 .

You are right, currently only the will topics are sent if disconnected unintentionally. I will add this as soon as I have fixed the total data conversion mess :D

closed 2024-03-20 by SeppPenner