SeppPenner/SparkplugNet #76: Interest to share with Eclipse Foundation?

scottmwyant, 2024-02-29:
Wondering if the owner of this repository has any interest or intention of sharing / donating this work to the [eclipse/tahu](https://github.com/eclipse/tahu) project?

scottmwyant, 2024-03-01:
I should have given a little more context. I am not affiliated with Eclipse Foundation in any formal capacity. I did join recently, only to submit a PR to fix something in their docs.

I _am_ a user of Sparkplug personally and professionally. I see a need for c# support from the Eclipse Tahu project.  I understand this is largely a matter of opinion, as 3rd party projects like this one are perfectly suitable from a technical standpoint.

What I'm asking you to do is open an issue in the Eclipse Tahu repository to understand their interest in transitioning this code to their ownership.  My thinking is that members of the community may be more inclined to contribute if the code were part of the Eclipse Tahu project. 

If you have no interest in any of this, that's perfectly fine, as I realize I'm a stranger on the internet. I don't know your motivations for this project and don't need to; I'm only asking the question.

I do hope my recommendation comes off as a compliment.

SeppPenner, 2024-03-03:
I asked the ecplise tahu people under https://github.com/eclipse/tahu/issues/364 :)

closed 2024-03-03 by SeppPenner