SeppPenner/SparkplugNet #78: Conversion issues with DateTime, Extension and PropertySetList

SeppPenner, 2024-03-03:
* DateTimes are in the wrong time zone somehow in the tests
* How to handle extension values?
* How to handle PropertySetList types?

shouidar, 2024-03-03:
I think this is occurring when casting between DateTime  and DateTimeOffset to use ToUnixMilliSeconds. I did some tests and it seems this can be avoided by using Universal Time all over the application (In fact Sparkplug Spec calls for UTC time on metric).
Making the following change in the _Metric.GetValue()_ Method should maintain correct conversion (Assuming dates are stored in UTC format, maybe additional checks should be performed in Metric.SetValue to ensure this)).

```C#
case VersionBDataTypeEnum.DateTimeArray:
                return this.ObjectValue.ConvertOrDefaultTo<ulong[]>().Select(v => DateTimeOffset.FromUnixTimeMilliseconds((long)v).UtcDateTime).ToArray();
```

In addition, I think the ByteValue Array in Test34 is Incorrect.

SeppPenner, 2024-03-04:
That might be the case with UTC. Need to check it. Test34 is the `DateTimeArray` type, I guess? Yes, this doesn't work properly yet. I still wanted to provide a release fixing some of the low code issues to give the update for the rest already to the people.

shouidar, 2024-03-04:
Done the following changes in my local branch, all seems to run smothly for time conversion:
1.In _Metric.GetValue_: used _UtcDateTime_ instead of _DateTime_
```C#
case VersionBDataTypeEnum.DateTimeArray:
                return this.ObjectValue.ConvertOrDefaultTo<ulong[]>().Select(v 
                    => DateTimeOffset.FromUnixTimeMilliseconds((long)v).UtcDateTime).ToArray();
```
2.Same In _ValuBaseVersionB Value_
```C#
        VersionBDataTypeEnum.DateTime => 
DateTimeOffset.FromUnixTimeMilliseconds(this.ObjectValue.ConvertOrDefaultTo<long>()).UtcDateTime,
```
And in _ValueBaseVersionB SetValue_ added:

```C#
else if (value is DateTimeOffset dateTimeOffset)
                {
                    this.ObjectValue = (ulong)dateTimeOffset.ToUnixTimeMilliseconds();
                    break;
                }
```
3. Finally in PayloadConverter Tests Initialized DateTimes as follow (UTC)

```C#
var dateTime = new DateTimeOffset(2020, 1, 1, 0, 0, 0, TimeSpan.Zero);
        var dateTime2 = new DateTimeOffset(2022, 1, 1, 0, 0, 0, TimeSpan.Zero);
        var dateTime3 = new DateTimeOffset(2023, 1, 1, 0, 0, 0, TimeSpan.Zero);
        var dateTime4 = new DateTimeOffset(2024, 1, 1, 0, 0, 0, TimeSpan.Zero);
```
And set the hard coded check values to :

```C#
BytesValue = [0x00, 0xE8, 0x66, 0x5E, 0x6F, 0x01, 0x00, 0x00,
                              0x00, 0x9C, 0xEF, 0x12, 0x7E, 0x01, 0x00, 0x00, 
                              0x00, 0xC8, 0xA0, 0x6A, 0x85, 0x01, 0x00, 0x00, 
                              0x00, 0xF4, 0x51, 0xC2, 0x8C, 0x01, 0x00, 0x00]
```
To avoid duplicate effort, please feel free to assign any task you may need assistance with.

SeppPenner, 2024-03-04:
@shouidar Sorry for confusing you... My tests were not properly done. The date time value checks where commented out. I added them again, now they don't work :D Still the issues with UTC / non UTC timezones. If you find the time and want to help me out, you can check these things.

shouidar, 2024-03-05:
@SeppPenner: Working on it.

shouidar, 2024-03-05:
This is happening. because when we supply a DateTime object to the Metric constructor, the DateTimeOffset(DateTime).ToUnixMilliSeconds() will convert the DateTime to UTC before calculating the milliseconds offset.
I think we should check if the DateTime supplied is in UTC and throw an exception to warn the user. Otherwise, the Metric conversion will result in a different long value depending on the system and location where the application is run.

SeppPenner, 2024-03-05:
> This is happening. because when we supply a DateTime object to the Metric constructor, the DateTimeOffset(DateTime).ToUnixMilliSeconds() will convert the DateTime to UTC before calculating the [...]

I would also favor the strategy with throwing an exception. It makes no sense to convert the date time to UTC by default just to tell the user that 'everything is okay' if the expected result is in local time in this case.

SeppPenner, 2024-03-14:
Conversion issues with DateTimes are fixed thanks to @shouidar. The others are still open, need to check it.

SeppPenner, 2024-03-22:
I will keep the tracking of this under https://github.com/SeppPenner/SparkplugNet/issues/89 now.

closed 2024-03-22 by SeppPenner