SeppPenner/SparkplugNet #80: Clients unable to parse published propertyset

oceanic76, 2024-03-04:
At birth or ndata the published  propertyset type of a SparkplugNet.VersionB.SparkplugNode cannot be parsed by MQTT.fx (Sparkplug Decoder) or Node-RED Sparkplug node . Tried 1.3.4 and 1.3.3 using both SparkplugMqttProtocolVersions and SparkplugSpecificationVersions. The objects appear okay up to the serialization and then I cannot tell if it is serialized correctly.

Code to build Metric object:
var ps = new PropertySet();
ps.Keys.Add("A");
var aValue = new PropertyValue(DataType.String, "testValue");
ps.Values.Add(aValue);
var m1 = new Metric("TestPropertySet", DataType.PropertySet, ps);

Node-RED: "Unable to decode Unknown message. the error is : TypeError: Cannot read properties of undefined (reading 'keys')"

SeppPenner, 2024-03-05:
Maybe a spelling issue in the library (or a Proto member name is set to invalid). I need to check this.

oceanic76, 2024-03-05:
The Sparkplug 3 specification document does not list PropertySet as a oneof data type for a Metric's value. The PropertySet (20) type is referenced for the "Properties" field. Using the Eclipse proto in an independent C# project, the clients also failed to parse the data. However, when the Metric's data type is set to string and a string value, then the message can be parsed by the client and the properties are present as children of the "properties" item. I have not figured out how to do it with SparkplugNet. 

This is not how I expected the PropertySet to work, however I also have not used it before.

{
	"metrics": [
		{
			"name": "testMetric",
			"dataType": "String",
			"properties": {
				"testKey": {
					"type": "String",
					"value": "testValue"
				}
			},
			"value": "test"
		}
	],
	"seq": 0
}

shouidar, 2024-03-06:
Yes, I looked at the eclipse tahu java implementation and the PropertySet is not part of the Metric value. It is used in the **properties** field (9) of the Metric itself as mentioned in the spec: ( _This is a PropertySet object associated with the metric for including custom key/value pairs of metadata associated with a metric. This is covered in the_). This is also true for version 2.2 version of the spec.

SeppPenner, 2024-03-22:
I can't reproduce this... By the way, this has nothing to do with `SparkplugMqttProtocolVersion` or `SparkplugSpecificationVersion` (Just because it was mentioned).

I have added a test for the use case, but can't reproduce: https://github.com/SeppPenner/SparkplugNet/commit/07a49ac98ee9d7a902535fd29cc29969731e8967.

SeppPenner, 2024-03-22:
@shouidar I don't want to disturb you, but since you seem to be the person to ask for that kind of `tricky` stuff, do you see where the issue is located?

shouidar, 2024-03-22:
As I mentioned above the property and propertySet are not value objects for then metric, they are part of the Metric  property fields (like name, alias etc...) on the spec there is a properties field that can take a propertyset.

I have already tested this on my local repo: I just added a Properties field to the Metric class and removed all the propertySet value from the value property.

```C#
public MetaData? MetaData { get; set; }

/// <summary>
/// 
/// </summary>
public PropertySet? Properties { get; set; } 
/// <summary>
/// Gets the value.
/// </summary>
public override object? Value => base.Value ?? this.GetValue();
```

This was decoded properly by MqttFx as discribed here: https://github.com/SeppPenner/SparkplugNet/issues/80#issuecomment-1978951522

I can submit a pull request for this if you like.

SeppPenner, 2024-03-22:
@shouidar Yeah, but it's only stored in the value in the internal metric for simplicity reasons. When converting to / from Proto, the converter uses the Properties field in the Proto metric. Check out my test for this, please. This should work properly if you ask me.

shouidar, 2024-03-22:
@SeppPenner  sorry did not check the last update. I will check and update here.

shouidar, 2024-03-22:
Still the same issue. What I meant is in the tahu sparkplug_b_c_sharp.proto and sparkplug_b.proto the Metric class DOES NOT have a value of type PropertySet. It has a properties member of type PropertySet which is not part of the oneof items of the Value meber.

In your generated ProtoBufPayload.cs there is a PropertySet added that why the tests succeeds. But other SparkPlug clients do not deserialize this member as a metric value.  They expect a separate Properties field of type PropertySet.

```c#
message Metric {

        optional string   name          = 1;        // Metric name - should only be included on birth
        optional uint64   alias         = 2;        // Metric alias - tied to name on birth and included in all later DATA messages
        optional uint64   timestamp     = 3;        // Timestamp associated with data acquisition time
        optional uint32   datatype      = 4;        // DataType of the metric/tag value
        optional bool     is_historical = 5;        // If this is historical data and should not update real time tag
        optional bool     is_transient  = 6;        // Tells consuming clients such as MQTT Engine to not store this as a tag
        optional bool     is_null       = 7;        // If this is null - explicitly say so rather than using -1, false, etc for some datatypes.
        optional MetaData metadata      = 8;        // Metadata for the payload
        optional PropertySet properties = 9;

        oneof value {
            uint32   int_value                      = 10;
            uint64   long_value                     = 11;
            float    float_value                    = 12;
            double   double_value                   = 13;
            bool     boolean_value                  = 14;
            string   string_value                   = 15;
            bytes    bytes_value                    = 16;       // Bytes, File
            DataSet  dataset_value                  = 17;
            Template template_value                 = 18;
            MetricValueExtension extension_value    = 19;
        }

        message MetricValueExtension {
            extensions  1 to max;
        }
    }
```

shouidar, 2024-03-23:
I think the Properties filed should be stored separately as a Metric can have both value and Properties. The purpose of the properties filed in my understanding of the spec and as I could see on the Tahu reference implementation is to provide additional details about the metric and **is not** a value by itself. 

I checked by adding the metric to list of metrics in the test application (see code bellow). 
```c#
private static readonly List<VersionBData.Metric> VersionBMetricsNode =
    [
        new VersionBData.Metric("temperatureNode", VersionBData.DataType.Float, 1.243f),
        new VersionBData.Metric("climateactiveNode", VersionBData.DataType.Boolean, true),
        new VersionBData.Metric("Test22", VersionBData.DataType.PropertySet, new VersionBData.PropertySet
        {
            Keys = ["Test1", "Test2"],
            Values =
            [
                new(VersionBData.DataType.Int8, 1)
                {
                },
                new(VersionBData.DataType.Int64, (long)2)
                {
                }
            ]
        })
    ];
```
The payload is serialized as expected with the Properties field set to the given PropertySet, and the DataType 20. as can be seen on screenshot below: (payload decoded by  [Protobufpall](https://www.protobufpal.com/).

[image]

But because other sparkplug clients do not expect a Metric of Datatype 20, they fail to decode it:
[image]

shouidar, 2024-03-23:
By adding the Properties field to the Metric class:
```c#
/// <summary>
    /// Gets or sets the Properties field.
    /// </summary>
    public PropertySet? Properties { get; set; }
```

And assigning it to the PropertySetValue during conversion (independently of the Metric value which is still converted separately depending on the metric DataType):

```c#
var protoMetric = new VersionBProtoBuf.ProtoBufPayload.Metric()
        {
            Alias = metric.Alias,
            DataType = (uint?)ConvertVersionBDataType(metric.DataType),
            IsHistorical = metric.IsHistorical,
            IsNull = metric.IsNull,
            IsTransient = metric.IsTransient,
            MetaData = ConvertVersionBMetaData(metric.MetaData),
            Name = metric.Name,
            Timestamp = metric.Timestamp,
            PropertySetValue = ConvertVersionBPropertySet(metric.Properties.ConvertOrDefaultTo<PropertySet>())
        };
```
Then creating the metric as follows:
```c#
new VersionBData.Metric("climateactiveNode", VersionBData.DataType.Boolean,null)
        {
            Properties = new VersionBData.PropertySet
            {
                Keys = ["ON", "OFF"],
                Values =
                [
                    new(VersionBData.DataType.Int8, 1)
                    {
                    },
                    new(VersionBData.DataType.Int8, 0)
                    {
                    }
                ]
            }
        }
```
It gets decoded properly by MqttFx:
[image]

SeppPenner, 2024-03-23:
So, the main problem is the data type... (Ok, and theoretically that a metric could have a value and a property set at the same time)...

I agree, we can change that. Since you already seem to have a working version, would you mind to put it into a PR @shouidar?

SeppPenner, 2024-03-23:
And thank you for the detailed analysis :)

shouidar, 2024-03-23:
> And thank you for the detailed analysis :)

[@spenner](https://github.com/SeppPenner) you are welcome. Submitted PR.

SeppPenner, 2024-03-24:
Fixed by https://github.com/SeppPenner/SparkplugNet/pull/90.

closed 2024-03-24 by SeppPenner