SeppPenner/SparkplugNet #83: Authentication with TLS Certificate

petrosinoE80, 2024-03-08:
Hello, thank you for the continuous bug fixes and updates provided. 

I was wondering if anyone has tried connecting via TLS certificates (I use these certificates to connect to a RabbitMQ).
I'll provide some lines of code that I use to attempt the connection without a username and password, but I'm getting an "Empty Stream" error which seems to be caused by the lack of TLS handshake. In addition, it signals that MqttClientOptionsBuilderTlsParameters is obsolete.

Any advice? 
Thank you very much

```C#
GetTlsParametersDelegate? getTlsParameters = useTLS == true ? new GetTlsParametersDelegate(BuildTls) : null;
```

```C#
sparkplugBNodeOpt = new SparkplugNodeOptions(..., getTlsParameters, ...);
...
sparkplugBNode.Start(sparkplugBNodeOpt);
```

```C#
[Obsolete]
private MqttClientOptionsBuilderTlsParameters BuildTls()
{
    string certificatePath = "certificate.pfx";
    string caCertificatePath = "caCertificate.crt";

    List<X509Certificate> certificateList =
    [
        new(certificatePathPFX,"password")
    ];

    var tlsParameters = new MqttClientOptionsBuilderTlsParameters
    {
        UseTls = true,
        Certificates = certificateList,
        AllowUntrustedCertificates = true, // TODO: remove in production
        IgnoreCertificateChainErrors = true, // TODO: remove in production
        IgnoreCertificateRevocationErrors = true, // TODO: remove in production
        CertificateValidationHandler = (context) => { return true; }
    };

    return tlsParameters;
}	
```

SeppPenner, 2024-03-12:
Maybe this is a question for @chkr1011, I need to check this.

SeppPenner, 2024-03-24:
The thing is that because Version A of Sparkplug is deprecated (but some still want to use it), I have marked the features as deprecated already. Since I have `TreatWarningsAsErrors` set, I need to disable these deprecation warnings. Then I don't see deprecated changes in the libraries I use (In this case MqttNet)... I need to check this, maybe just updating helps.

petrosinoE80, 2024-03-25:
Thank you for the updates provided!

chkr1011, 2024-03-26:
@SeppPenner Do you still need my input here?

SeppPenner, 2024-03-26:
> @SeppPenner Do you still need my input here?

Are there examples in the MQTTnet repo? If yes, then I don't neet further information, I guess.

SeppPenner, 2024-03-27:
@chkr1011 Like this, I guess:

1. Remove proxy options.
2. Rework WebSocket options with class `MqttClientWebSocketOptions` in the options and then rework the calls to:

**WebSocketServer options:**

```csharp
if (this.Options.MqttWebSocketOptions is null)
{
    builder.WithTcpServer(this.Options.BrokerAddress, this.Options.Port);
}
else
{
    builder.WithWebSocketServer(o =>
        o.WithCookieContainer(this.Options.MqttWebSocketOptions.CookieContainer)
        .WithCookieContainer(this.Options.MqttWebSocketOptions.Credentials)
        .WithProxyOptions(this.Options.MqttWebSocketOptions.ProxyOptions)
        .WithRequestHeaders(this.Options.MqttWebSocketOptions.RequestHeaders)
        .WithSubProtocols(this.Options.MqttWebSocketOptions.SubProtocols)
        .WithUri(this.Options.BrokerAddress)
        .WithKeepAliveInterval(this.Options.MqttWebSocketOptions.KeepAliveInterval)
        .WithUseDefaultCredentials(this.Options.MqttWebSocketOptions.UseDefaultCredentials)
    );
}
```

3. Rework TLS options from `GetTlsParameters` to class `MqttClientTlsOptions` in the options and then rework the calls to:

**TLS options:**

```csharp
if (this.Options.MqttTlsOptions is not null)
{
    builder.WithTlsOptions(this.Options.MqttTlsOptions);
}
else
{
    builder.WithTlsOptions(o => o.UseTls());
}
```

SeppPenner, 2024-03-27:
@chkr1011 Just one thing, can you have a quick look over the changes, please? https://github.com/SeppPenner/SparkplugNet/commit/dfc03bb67f89dcae1f58aa52a4dbf88f7698ab0d. Thanks :)

chkr1011, 2024-04-01:
Sorry but I am a little bit lost here. Which changes do you want me to review? All of them? Or do you want me to have a look at a certain change? :confused:

SeppPenner, 2024-04-02:
> @chkr1011 Like this, I guess:
> 
>     1. Remove proxy options.
> 
>     2. Rework WebSocket options with class `MqttClientWebSocketOptions` in the options and then rework the calls to:
> 
> 
> **WebSocketServer options:**
> 
> ```cs
> if (this.Options.MqttWebSocketOptions is null)
> {
>     builder.WithTcpServer(this.Options.BrokerAddress, this.Options.Port);
> }
> else
> {
>     builder.WithWebSocketServer(o =>
>         o.WithCookieContainer(this.Options.MqttWebSocketOptions.CookieContainer)
>         .WithCookieContainer(this.Options.MqttWebSocketOptions.Credentials)
>         .WithProxyOptions(this.Options.MqttWebSocketOptions.ProxyOptions)
>         .WithRequestHeaders(this.Options.MqttWebSocketOptions.RequestHeaders)
>         .WithSubProtocols(this.Options.MqttWebSocketOptions.SubProtocols)
>         .WithUri(this.Options.BrokerAddress)
>         .WithKeepAliveInterval(this.Options.MqttWebSocketOptions.KeepAliveInterval)
>         .WithUseDefaultCredentials(this.Options.MqttWebSocketOptions.UseDefaultCredentials)
>     );
> }
> ```
> 
>     3. Rework TLS options from `GetTlsParameters` to class `MqttClientTlsOptions` in the options and then rework the calls to:
> 
> 
> **TLS options:**
> 
> ```cs
> if (this.Options.MqttTlsOptions is not null)
> {
>     builder.WithTlsOptions(this.Options.MqttTlsOptions);
> }
> ```

@chkr1011 Just the changes in the comment (above), please :)

SeppPenner, 2024-07-02:
I guess, this is done.

closed 2024-07-02 by SeppPenner