SeppPenner/SparkplugNet #84: Not able to publish element values in DataSet datatype values

ankitagojiya, 2024-03-11:
Hello @SeppPenner ,

In SparkplugNet version 1.3.4 , not able publish elements values in DataSet because from my analysis i think there is an error in method "ConvertVersionBDataSetValue" in PayloadConverter class , We are comparing types index(Which will be always 0 , 1 , 2 not datatypes) instead of dataSetValue datatype that's why it will always throw "ArgumentOutOfRangeException" error for any datatype in DataSetValue.
[image]

I think we have to use below condition
[image]

It was working fine in SparkplugNet version 1.3.2 pls refer below image.
[image]

@SeppPenner It will be good if we resolve this ASAP , because I am block with this.

ankitagojiya, 2024-03-11:
I have found another way of using this without above changes , is this the correct way to use it ?

[image]
Do we need to pass number of elements in " Types = new uint[3]" ?
There is any specific requirement to pass data everytime like this bcz it is not used anywhere else rather than looping ?

SeppPenner, 2024-03-12:
> I have found another way of using this without above changes , is this the correct way to use it ? [...]

This is the way, yes. Still, I agree that this should be improved.