SeppPenner/SparkplugNet #86: Not able to access StringValue, IntValue, TemplateValue but all the details will be now come under the value section.(Version 1.3.4)

malavvvakharia, 2024-03-12:
Hello @SeppPenner ,

Is there any enhancement or specific reason for not giving the data in IntValue, StringValue, or any other type? (It is properly working before the version release of 1.3.4)

[image]

According to the below SS is not able to access the StringValue,IntValue, etc.:

[image]

SeppPenner, 2024-03-12:
Basically, the internal metrics have changed in version 1.3.4 which means that there is only `Value` that is filled according to the data type specified. So, if you input / receive a metric with type `String`, the value will be a `string` value, but its generic type is `object` for convenience issues / simplification.

You can try `metricValue.Value as string stringValue` or something like

```csharp
if (metricValue.Value is string stringValue)
{
    // Do something with the string value from 'stringValue'
}
```

This is not a bug, but intended behaviour as I wanted to have a generic API that works with all data types equally.

closed 2024-03-12 by SeppPenner

malavvvakharia, 2024-03-14:
Thanks for the information @SeppPenner .
I am confused as it is not mentioned in the release version of 1.3.3 and 1.3.4.
(https://github.com/SeppPenner/SparkplugNet/releases/tag/1.3.4)