SeppPenner/SparkplugNet #88: PropertySet

petrosinoE80, 2024-03-22:
Thank you for the continuous repository updates. 

I would need to understand how to add a PropertySet to a Metric (if it is possibile), as Metric.Properties is no longer exposed. 

Thank you very much in advance.

SeppPenner, 2024-03-22:
This doesn't work at the moment, possible duplicate of https://github.com/SeppPenner/SparkplugNet/issues/80.

SeppPenner, 2024-03-22:
This is one of the things I try to fix next (As well as PropertySetList which doesn't work yet, too).

SeppPenner, 2024-03-22:
The information I gave before was wrong. I have added a new test under https://github.com/SeppPenner/SparkplugNet/blob/07a49ac98ee9d7a902535fd29cc29969731e8967/src/SparkplugNet.Tests/Payloads/SparkplugPayloadConverterTestVersionB.cs#L1848, but I didn't find an issue.

However, you should be able to add the property set values like this (Same as the test):

```csharp
var metrics = new List<VersionBData.Metric>
        {
            new("Test22", VersionBData.DataType.PropertySet, new VersionBData.PropertySet
                {
                    Keys = ["Test1", "Test2"],
                    Values =
                    [
                        new(VersionBData.DataType.Int8, (sbyte)1)
                        {
                        },
                        new(VersionBData.DataType.Int64, (long)2)
                        {
                        }
                    ]
                }, timestamp)
            {
                Alias = 22,
                IsHistorical = true,
                IsTransient = true
            }
        };
```

I hope this helps. If you find another issue with the property set, please follow Follow https://github.com/SeppPenner/SparkplugNet/issues/80. I try to resolve the issue there.

closed 2024-03-22 by SeppPenner