SeppPenner/SparkplugNet #96: Negative integer set value

petrosinoE80, 2024-04-16:
Hey there, while testing the NuGet 1.3.8 update, I noticed that I'm no longer receiving messages with negative integers in the MQTT broker side. 

Here's an example: 

```csharp

Metric metric = new()
{
	Name = "Test",
	Timestamp = timestamp_var
};

Int16 negativeInt = -1;

metric.SetValue(DataType.Int16, negativeInt);

```

In previous versions of the NuGet, I had a workaround where I casted the data to Float, but now I'm still not getting anything. Is this a known issue? Any possible solutions? 

Thanks in advance for all the efforts you put into improving this fantastic NuGet.

SeppPenner, 2024-04-22:
I guess, this can't be fixed with version 3 anymore and is a known issue within Sparkplug still: https://github.com/eclipse/tahu/issues/71...

faramart, 2024-05-09:
Hi, thank you very much for addressing this issue. I'm also struggling with this right now and I'm trying to find a possible workaround.

When I try to send -1 int on version 1.3.8, I get OverflowException.
```
System.OverflowException: Value was either too large or too small for a UInt32.
   at System.Convert.ThrowUInt32OverflowException()
   at System.Int32.System.IConvertible.ToUInt32(IFormatProvider provider)
   at System.Convert.ChangeType(Object value, Type conversionType, IFormatProvider provider)
   at System.Convert.ChangeType(Object value, Type conversionType)
   at SparkplugNet.Core.Extensions.DataTypeExtensions.ConvertTo[T](Object objValue)
   at SparkplugNet.Core.Extensions.DataTypeExtensions.ConvertOrDefaultTo[T](Object objValue)
   at SparkplugNet.VersionB.PayloadConverter.ConvertVersionBMetric(Metric metric)
   at System.Linq.Enumerable.SelectListIterator`2.ToList()
   at System.Linq.Enumerable.ToList[TSource](IEnumerable`1 source)
   at SparkplugNet.VersionB.PayloadConverter.ConvertVersionBPayload(Payload payload)
   at SparkplugNet.Core.Messages.SparkplugMessageGenerator.GetSparkplugNodeDataB(SparkplugNamespace nameSpace, String groupIdentifier, String edgeNodeIdentifier, IEnumerable`1 metrics, Int32 sequenceNumber, DateTimeOffset dateTime)
   at SparkplugNet.Core.Messages.SparkplugMessageGenerator.GetSparkplugNodeDataMessage[T](SparkplugNamespace nameSpace, String groupIdentifier, String edgeNodeIdentifier, IEnumerable`1 metrics, Int32 sequenceNumber, Int64 sessionNumber, DateTimeOffset dateTime)
   at SparkplugNet.VersionB.SparkplugNode.PublishMessage(IEnumerable`1 metrics)
   at SparkplugNet.Core.Node.SparkplugNodeBase`1.PublishMetrics(IEnumerable`1 metrics)
```
As far as I can tell, the correct approach for sending negative values with the current sparkplug specification is the following:

- Set correct  Metric.datatype field e.g. Int32
- Cast the INT32 Value, for example “-32” to UINT32, e.g. “4294967264”
- Add the converted value as metric.int_value
- Send via MQTT

I have seen this described [here](https://community.hivemq.com/t/sparkplug-how-to-deal-with-negative-signed-integers/463) and [here](https://github.com/eclipse-sparkplug/sparkplug/issues/114#issuecomment-914247099) and have tested it using Node-RED. Looks like it is working well. 

Could this logic be part of SparkplugNet?

SeppPenner, 2024-05-13:
It seems like C# doesn't convert the types properly with `Convert`...

SeppPenner, 2024-07-02:
This is basically just:

```csharp
int intValue = -32;
uint uintValue = unchecked((uint)intValue);
Console.WriteLine(uintValue);
```

SeppPenner, 2024-07-02:
I have updated the tests and nearly fixed this in https://github.com/SeppPenner/SparkplugNet/commit/d3ff29bd4c26c35c8cf963955a7105349177e0c1. However, the conversions back from https://github.com/SeppPenner/SparkplugNet/blob/d3ff29bd4c26c35c8cf963955a7105349177e0c1/src/SparkplugNet/VersionB/PayloadConverter.cs#L219 and https://github.com/SeppPenner/SparkplugNet/blob/d3ff29bd4c26c35c8cf963955a7105349177e0c1/src/SparkplugNet/VersionB/PayloadConverter.cs#L224 don't work yet properly. Feel free to share your thoughts, I don't see the issue here yet...

SeppPenner, 2024-07-02:
This code is crazy now: https://github.com/SeppPenner/SparkplugNet/blob/d3ff29bd4c26c35c8cf963955a7105349177e0c1/src/SparkplugNet/Core/Extensions/DataTypeExtensions.cs :D

closed 2024-07-02 by SeppPenner

faramart, 2024-08-16:
Thank you for the changes. ❤ It is working ok for my use case and I'm able to send negative values without any issue.