SeppPenner/SparkplugNet #98: Not able to deserialize the sparkplug metric via System.Text.Json or  Newtonsoft.Json.

malavvvakharia, 2024-05-21:
For example below is the one example of JSON which I need to deserialize to List<Metric>:

Example:
`[
  {
    "Name": "configuration/ABC",
    "Alias": null,
    "Timestamp": null,
    "IsHistorical": null,
    "IsTransient": null,
    "IsNull": false,
    "MetaData": null,
    "Properties": null,
    "Value": {
      "Version": "",
      "Metrics": [
        {
          "Name": "product",
          "Alias": null,
          "Timestamp": null,
          "IsHistorical": null,
          "IsTransient": null,
          "IsNull": false,
          "MetaData": null,
          "Properties": null,
          "Value": "ABC",
          "DataType": 12
        }
      ],
      "Parameters": [
        
      ],
      "TemplateRef": "",
      "IsDefinition": null
    },
    "DataType": 19
  }
]`

C# code:

`var metric = JsonConvert.DeserializeObject<List<Metric>>(metrics);
  var metric = System.Text.Json.JsonSerializer.Deserialize<List<Metric>>(metrics);
`

Now after deserialize, it will give value as null.

[image]

SeppPenner, 2024-07-02:
What is `metrics` and where does it come from? The library is just there to serialize / deserialize data to / from Protobuf and to conveniently work with Sparkplug / Protobuf logic, but isn't optimized for JSON serialization (And I guess with the current model, this isn't even possible)...

SeppPenner, 2024-07-02:
In my opinion, this shouldn't be part of the library as e.g. Annotations for System.Text.Json / Newtonsoft.Json would pollute the data models while the use case in the library is only Protobuf, not JSON...

SeppPenner, 2024-07-02:
I might try something though. (Maybe extension packages).