SeppPenner/SparkplugNet #101: HostApplication Required Known Metrics

ankitagojiya, 2024-07-11:
Hello @SeppPenner 

In Sparkplug's latest version 1.3.10, I am encountering the same issue as documented in issue #58, which was resolved in version 1.3.2. According to the specifications, the Host Application should not need to know in advance what metrics can be received from the Edge Node. However, in the latest version, we are required to predefine the incoming metrics in the Host Application. This is because the Host Application filters the incoming metrics from the Edge Node and only then invokes the NBIRTH/NDATA events. This behavior is not correct as per the specifications.

[image]
[image]

giovanni-ferrari, 2024-07-30:
i'm currently affected by this issue as well, that force me to stick to 1.3.2 version. 
I see there's a PR to fix it. 
Is there any chance to know when next release is planned? 
Thanks a lot

gojiyaankita, 2024-09-05:
Hello @SeppPenner 
Is there any updates regarding this issue ?

adityashahazilen, 2024-09-24:
Hello @SeppPenner 

When are you planning to look into this issue? As this is blocker for us and our application cannot deploy on production with this issue. Please take this on priority.

Thanks