SeppPenner/SparkplugNet #108: Question: Why is UInt32 transferred as LongValue?

heppth, 2024-08-13:
I did some tests with SparkplugNet and came across some ambiguities.

A UInt32 is interpreted by the server as a 0 because the server implementation evaluates IntValue of the protobuf message.

The mapping between SparkplugB data types and Protobuf data types is somewhat confusing. In this case, however, it is relatively clear to me: unsinged Int32 should become unsigned Int32, as described in the specification:
https://github.com/eclipse-sparkplug/sparkplug/blob/e45b61002a2a0cacb4f11fa4be88670435ccfcb0/specification/src/main/asciidoc/chapters/Sparkplug_6_Payloads.adoc?plain=1#L899-L902

But your implementation writes the value into LongValue:
https://github.com/SeppPenner/SparkplugNet/blob/758b561206807f8c46ee11e6594054ce3ca75ff8/src/SparkplugNet/VersionB/PayloadConverter.cs#L221-L225

Now my questions:
- Does the server need to be able to handle it if there is a UInt32 value in the LongValue field?
- Is there a reason for this?

SeppPenner, 2024-12-10:
I guess, this was just one stupid failure on my side...

closed 2024-12-10 by SeppPenner

tszpakowicz, 2025-04-01:
Please note that this change breaks unit test `TestConvertVersionBPayloadToProto`